RoPieee/RoPieee #4: [Feature]: Auto suspend for Topping E30

DeLub, 2025-08-26:
### Contact Details

alex@lubberts.org

### What's the idea?

Hi there!

After experimenting with audio on my Pi, I’ve fully transitioned to Roon and your fantastic Ropieee solution. (Thanks a lot!) However, I’ve encountered an issue where auto suspend doesn’t work with my Topping E30 DAC. Could you please add support for the E30?

Before using Ropieee, I managed to get it working with this udev rule:

`ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="152a", ATTR{idProduct}=="8750", TEST=="power/control", ATTR{power/control}="auto"`

I hope this helps.

Thanks in advance!

spockfish, 2025-08-27:
Curious as why the auto suspend functionality from RoPieee does not work. 
Can you send me feedback?

DeLub, 2025-08-27:
When I wanted to send feedback I saw an update being available. I installed it, checked again, and now it's working... Nothing else changed.

closed 2025-08-29 by spockfish