RoPieee/RoPieee #7: [Feature]: Possibility to limit or set card volume for Plexamp

moonbit5, 2025-08-31:
### What's the idea?

Hi thanks for the great distro.

Is there any way to limit the master volume for a Plexamp player? Kind of like in Lyrion server where the master vol can be set.

Currently the volume of Plexamp is incredibly loud and could do with some kind of safety feature here if possible? 

Thank you!

spockfish, 2025-08-31:
Of course technically it's possible, so maybe my choice of words was not correct.
It's more that I'm not convinced that this feature belongs in RoPieee: I would expect that Plex has this kind of feature.

Are you using a USB DAC or a HAT?

moonbit5, 2025-08-31:
I'm using a hifiberry amp HAT

oakbrad, 2025-11-08:
Could this potentially be configured in the Audio HAT settings? I'm also using a Hifiberry (DAC+ Zero) and for whatever reason it is very loud (like 3-4x!) relative to the other devices I have plugged into my amp. 

I haven't tried any services besides Plexamp but I don't think Plex has a way to set a master volume - it would be controlled by the hardware/OS I believe.