RoPieee/RoPieee #9: [Bug]: Sqeezelite not working in 2025.08.1

MichaelSch, 2025-09-01:
### What happened?

I have an issue after updating from 2025.08 to 2025.08.1. None of my devices is shown in my Lyrion Music Server anymore. Other services are working as expected. I tried to reconfigure Squeezelite, but it does not show up.

Feedback: 212dd0a1c9e9a20f

### Version

2025.08.1 (Default)

### Hardware

Raspberry Pi 5

### Additional information

```shell

```

MichaelSch, 2025-09-01:
Already sent - see initial report. Here's a new one:

spockfish, 2025-09-01:
Can someone validate this with a forced IP address for the server? you can configure this in the RoPieee website (last option on the Squeezelite tab). 

If I do that on unit it works.

patman62, 2025-09-01:
@spockfish That fixed both of my devices. Thanks.

MichaelSch, 2025-09-01:
I can also confirm that setting the server‘s IP fixes the issue.

spockfish, 2025-09-07:
I've released a new beta (2025.08.2) that should solve this.

closed 2025-09-09 by spockfish