RoPieee/RoPieee #17: Hifiberry dac nicht erkannt

Martinmoe59, 2025-09-19:
### Contact Details

martinmoe59@gmail.com

### What happened?

Ropieee erkennt meinen Dac hifiberry +ADC nicht

### Version

2025.08.2 (Default)

### Hardware

Raspberry Pi 3

### Additional information

```shell

```

Martinmoe59, 2025-09-24:
> @Martinmoe59<https://github.com/Martinmoe59>
> wie hast du es zum Laufen gebracht.
> unfortunately killed the Roipeee system trying jackd out 👎

Hallo
Ich hatte einen Fehler gehabt beim Anschließen und ich hatte den Hifiberry in Roon nicht aktiviert sondern nur auf die LRDs des Hifiberry geachtete, die leuchtinstalliert isten aber erst wenn alles komplett

Martin Möllinger
Kandelstraße 10
79331 Teningen
Mobil 015222773696
***@***.***

spockfish, 2025-09-24:
But again, would be great if someone with this HAT can send me feedback from the RoPieee webpage. This helps me in integrating it properly.