RoPieee/RoPieee #19: [Bug]: Spotify Connect

follango-coder, 2025-09-24:
### Contact Details

gfollan1@nycap.rr.com

### What happened?

Still running into the same issue with Spotify connecting to the Ropieee bridge and then disconnecting immediately with the latest Beta "RoPieee 2025.10 (2760) [BETA]". This is on both bridges I own V4 and V5. Originally started with 2025.08.2.

### Version

2025.08.2 (Default)

### Hardware

Raspberry Pi 5

### RoPieee Feedback ID

d1c14b6cb23c85e5

### Additional information

```shell

```

spockfish, 2025-09-25:
The logs show that the player can't find certain data...

Hmmm.... can you try selecting a different track/album? Something popular?

Just want to verify if this might be the case.

follango-coder, 2025-09-25:
Seems to play various Spotify source material fine on the Android phone,
but exhibits the same behaviour when switching the output to the Raspberry
Pi bridge. Connects and immediately halts and switches back to the phone
output. I'll charge up an old phone to see if that changes anything.  Let
me know if there's any other info I can provide. Thanks

follango-coder, 2025-09-25:
Same issue with an alternate phone. I can cast to my desktop or AudioPro,
but not to Ropieee. Plexamp on Ropieee seems to be working fine, so
probably not a network configuration problem.

spockfish, 2025-09-25:
> Seems to play various Spotify source material fine on the Android phone,
but exhibits the same behaviour when switching the output to the Raspberry
Pi bridge

Can you do that with various material, and after some attempts ( 4-5 or so) send me feedback again?

spockfish, 2025-09-27:
I've pushed out another beta release, that has the latest Spotify player.

follango-coder, 2025-09-27:
Re-tested at 12:15 EDST. Same symptoms, no change in behavior. Feedback sent 899538da8fec9062, thanks.

spockfish, 2025-09-28:
On RoPieee.

follango-coder, 2025-09-29:
That works, toggling the audio output off and on for Plexamp reproduces the
issue. Plexamp output off, Spotify works. Plexamp output enabled, Spotify
breaks